chipsalliance/Cores-VeeR-EH1 #5: Difference behavior for system Bus Access in dbg.sv and RISC-V External Debug Support Version 0.13.2

ycyang0508, 2019-07-12:
In section 3.12.23 of "RISC-V External Debug Support Version 0.13.2", write "sbdata0" operation will trigger the system bus write access operation. But in dbg.sv, the write or read access operation is decided by sbcs_reg[15] or sbcs_reg[20] and this seems don't match with the specification.

ZvonimirBandic, 2019-07-12:
Thanks for moving from Western Digital GitHub. The issue has been assigned, we will respond by early next week.

tmw-wdc, 2019-07-16:
Hi Ya-Chau,

Thank you very much for bringing this issue to our attention.  We can confirm that this is a bug in our design and will be fixed soon.

Kind regards,
Thomas

ycyang0508, 2019-07-16:
glad to learn something new from SweRV and can contribute something to it at the same time.

jrahmeh, 2019-08-13:
Fixed in version 1.2.

closed 2019-08-13 by jrahmeh