chipsalliance/Cores-VeeR-EH1 #9: SweRV build breaks on latest Verilator

toddstrader, 2019-07-26:
Just a heads-up.  I'm working on head-to-head Verilator CI with external projects and found that the latest Verilator tosses some width warnings on the SweRV build:
https://www.veripool.org/issues/1483-Verilator-Make-verilator_ext_tests-head-to-head

I'm sure this isn't urgent, so I can touch things up but I won't be able to get to it for a week or so.  I thought I'd post here in case someone cares before then.

ZvonimirBandic, 2019-07-26:
Todd, we will look into this as well. When/if we eliminate SweRV being issue, we will let you know here.

aprnath, 2019-08-10:
These are warnings in the genvar procedures. 

```
   for (genvar i=1; i<40; i++) begin
      assign error_mask[i-1] = (ecc_check[5:0] == i);
   end
```
I am not sure these should be flagged or cause build failure.

toddstrader, 2019-08-10:
Yeah, sorry I should have circled back here.  This is being tracked on Verilator now with the intention of reverting to the old behavior.  SweRV shouldn't need to change.

https://www.veripool.org/issues/1487-Verilator-New-WIDTH-warnings-on-genvars

closed 2019-08-10 by toddstrader