chipsalliance/Cores-VeeR-EH1 #11: SweRV VLNV

olofk, 2019-07-31:
As part of packaging the SweRV EH1 core, I'm trying to figure out what would be the correct VLNV identifier. For those unaware of VLNV, it's an classification system for IP cores, stemming from the IP-XACT standard and stands for Vendor Library Name Version.

Vendor would be `chipsalliance.org` and version would be `1.1` in this case, but I'm not sure about the library and name parts. My two proposals are

    library = cores
    name = SweRV_EH1

or 

    library = SweRV
    name = EH1

The first one is based on that it belongs to the chipsalliance Cores library. The second one is because I have seen SweRV itself being referred to as a library of processors.

Anyway, both are fine with me, but I would like some input from the designers and those with a roadmap

robertgolla, 2019-08-17:
We also think both are fine but have a slight preference for the 1st option:  library = cores, name = SweRV_EH1

olofk, 2019-08-19:
Thanks. I'll go with that

closed 2019-08-19 by olofk