chipsalliance/Cores-VeeR-EH1 #14: Issues to enable DMA for SweRV

rajoogupta, 2019-09-04:
Hi,
I am trying to enable DMA for SweRV. I did following changes in **testbench/tb_top.sv**

   1.added following signal
    logic [31:0] dma_haddr;
    logic [2:0] dma_hburst;
    logic [3:0] dma_hprot;
    logic [2:0] dma_hsize;
    logic [1:0] dma_htrans;
    logic dma_hmastlock;
    logic dma_hwrite;
    wire dma_hresp_out;

   2. port mapping
    // DMA Slave
    //---------------------------------------------------------------
    .dma_haddr **('0) replaced with** ( dma_haddr ),
    .dma_hburst **('0) replaced with** ( dma_hburst ),
    .dma_hmastlock **('0) replaced with** ( dma_hmastlock),
    .dma_hprot **('0) replaced with** ( dma_hprot ),
    .dma_hsize **('0) replaced with** ( dma_hsize),
    .dma_htrans **('0) replaced with** ( dma_htrans),
    .dma_hwrite **('0) replaced with** ( dma_hwrite ),
    .dma_hwdata **('0) replaced with** ( dma_hwdata ),

        .dma_hrdata          ( dma_hrdata    ),
        .dma_hresp          **(dma_out)   **replaced with****   ( dma_hresp_out   ),
        .dma_hsel            ( 1'b1            ), 
        .dma_hreadyin        ( dma_hready_out  ),
        .dma_hreadyout       ( dma_hready_out  ),

  3.  added instance
    ahb_sif **o_ahb_dma** (

    .HWDATA(dma_hwdata),
    .HCLK(core_clk),
    .HSEL(1'b1),
    .HPROT(dma_hprot),
    .HWRITE(dma_hwrite),
    .HTRANS(dma_htrans),
    .HSIZE(dma_hsize),
    .HREADY(dma_hready),
    .HRESETn(reset_l),
    .HADDR(dma_haddr),
    .HBURST(dma_hburst),

    .HREADYOUT(dma_hready),
    .HRESP(dma_hresp),
    .HRDATA(dma_hrdata)

    );

after all , ran verilator and got hello world but **DMA is not enable** as ICCM and DCCM donot getting any data.
Here is GTKwave for reference

[image]

Please help me, to enable DMA for SweRV

Thanks & Regards
Rajoo

aprnath, 2019-09-06:
Hi Rajoo,

 Looking a your waveform, I notice a couple of things:

- Your dma_bus_clk_en is 0. This should be 1 to enable the DMA bus interface clocking. See the PRM for details on how the clock enables control clock ratios. A 0 effectively disables the clock.

- I do not see any DMA read/swrite initiated in the portion of the waveform shown. Your testbench bfm/DMA master must initiate read/write sequences on the bus. The core is a slave and will respond.

-Ajay

rajoogupta, 2019-09-06:
Hi Ajay,
             I set **dma_bus_clk_en** to 1 as you suggested (also referred  PRM) but DMA is not enable.
Here snapshot of DMA master signal.

[image]

Thanks & Regards 
Rajoo

aprnath, 2019-09-06:
Please show me a snapshot of waveform where your testbench  AHB master is initiating a read or write on the dma* AHB bus (assuming you are using the AHB flavor of the interface and not AXI).

The DMA port is a slave port and an external agent has to initiate the activity.

rajoogupta, 2019-09-06:
Hi Ajay,
             Please find snapshot of AHB master initiating a read or write on the dma AHB bus.
[image]

Previously there is no code in testbench to initiating a read or write on the dma AHB bus.
so added following code in testbench as I mentioned in my issue post.

[image]

Thanks & Regards 
Rajoo

rajoogupta, 2019-09-07:
Hi, 
      I am trying to enable DMA for SweRV. Inside design , there is a DMA controller which work as master for DMA. 
if anybody enable DMA , Please help me to enable it.

Thanks & Regards
Rajoo

aprnath, 2019-09-07:
Hi Rajoo,

The SweRV core has a DMA slave port, not a DMA master port. You need an external master to pull/push data from/to the core. 

Ajay

milk0920, 2019-09-12:
Hi , if we want to use the iccm ，Can I simply edit the link file， And let the cpu core to automatically download the code from external rom/ram/flash, is that possible, if we do not have the dma master design, but we still want to run code in iccm

closed 2020-02-11 by aprnath