chipsalliance/Cores-VeeR-EH1 #19: Questa support in Makefile for simulations

svenka3, 2019-10-01:
I tried running SweRV with Mentor Graphics's Questa simulator. I see testbench/flist.questa file-list, but Makefile didn't have targets for the same. I added the same and then found few minor issues in flist.questa, I have fixed and marked them with "// VW" for your kind review. 

Please see if you can incorporate these changes and/o validate the same at your end.

Thanks
Srini

[Makefile.txt](https://github.com/chipsalliance/Cores-SweRV/files/3674018/Makefile.txt)
[flist.questa.txt](https://github.com/chipsalliance/Cores-SweRV/files/3674020/flist.questa.txt)

agrobman, 2020-02-12:
You can have your own makefile/script/TB to run swerv with any simulator we provide just very basic way to show that design is alive. We checked that design can be compiled with questa, before release.

aprnath, 2020-02-20:
Please see updates in new release. Reopen if needed.

closed 2020-02-20 by aprnath