chipsalliance/Cores-VeeR-EH1 #28: Software breakpoints with OpenOCD + GDB

cstewart-ust, 2019-12-02:
When connecting OpenOCD I get the following warning:

Info : datacount=2 progbufsize=0
Warn : We won't be able to execute fence instructions on this target. Memory may not always appear consistent. (progbufsize=0, impebreak=0)

With icache enabled is there any way of reliably setting a SW breakpoint? Build options via swerv.config? I imagine this would require a fence instruction to be issued to flush cache but there's no program buffer to do so...

The custom CSR (0x7C4) seems that it would work via 'expose_csr' on OCD but initial results show inconsistent behavior

aprnath, 2020-02-11:
Yes, you need to use the custom DMST (0x7c4) CSR to force a fence.i before returning from debug mode, if you have used the system bus to update cacheable memory.

Could you be more specific about the "inconsistent behavior" ? Whats your setup and flow? Please provide your use case and steps so we can try and understand if it is a usage issue or a design issue.

agrobman, 2020-04-04:
OpenOCD out of box doesn't support DMST register, you'll need write it manually with debugger command every time you setting SW Breakpoint, or patch openOCD, including the register write into SW Breakpoint setting/removing functions.

closed 2020-04-13 by aprnath

JanMatCodasip, 2020-09-22:
A note in case anyone else runs into the same issue - 

With the present day SweRV 1.8+ and [riscv-openocd](https://github.com/riscv/riscv-openocd/), the **preferred solution** is to enable abstract memory access. That will properly populate the SW breakpoint even when ICache is present. Add the following line to your OpenOCD configuration file:

```
riscv set_mem_access abstract
```

For robustness, you may also have OpenOCD clear the ICache every time the processor is resumed from the debug mode. To do so, add this to your OpenOCD configuration:

```
# Expose custom SweRV CSR dmst (CSR 0x7c4 == 1988)
riscv expose_csrs 1988

proc swerv_eh1_execute_fence {} {
    # Execute fence + fence.i via "dmst" register
    reg csr1988 0x3
}

# Configure events hooks in OpenOCD to execute Fence + Fence.i when resuming
# the processor from the debug mode:

$_TARGETNAME configure -event resume-start {
    swerv_eh1_execute_fence
}

$_TARGETNAME configure -event step-start {
    swerv_eh1_execute_fence
}

```