chipsalliance/Cores-VeeR-EH1 #35: running with stack in RAM

monniaux, 2020-02-18:
I have tried running on the "verilated" simulated core a simple program making a function call but hit a snag.

The program is

```
#include "defines.h"

#define STDOUT 0xd0580000

.section .text
.global _start
_start:
    la   sp, _sp
	
    // Clear minstret
    csrw minstret, zero
    csrw minstreth, zero

    // Set up MTVEC - not expecting to use it though 
    li x1, RV_ICCM_SADR
    csrw mtvec, x1
 
 
    // Enable Caches in MRAC
    li x1, 0x55555555
    csrw 0x7c0, x1
 
    call main
	
// Write 0xff to STDOUT for TB to termiate test.
_finish:
    li x3, STDOUT
    addi x5, x0, 0xff
    sb x5, 0(x3)
    beq x0, x0, _finish
.rept 100
    nop
.endr

	.align	1
	.globl	main
	.type	main, @function
main:
	addi	sp,sp,-256
	sw	ra,252(sp)
	sw	s0,248(sp)

.rept 7
	nop
.endr
	
	lw	ra,252(sp)
	lw	s0,248(sp)
	addi	sp,sp,256
	jr	ra
	.size	main, .-main
```

The link file is
```
OUTPUT_ARCH( "riscv" )
ENTRY(_start)

MEMORY
{
	RAM  (wxa!ri) : ORIGIN = 0x00000000, LENGTH = 64K
}

/*----------------------------------------------------------------------*/
/* Sections                                                             */
/*----------------------------------------------------------------------*/

SECTIONS
{
  __stack_size = DEFINED(__stack_size) ? __stack_size : 4K;

  .text.init	:
  {
	*(.text.init)
	. = ALIGN(8);
  } > RAM

  .text :
  {
    *(.text.unlikely .text.unlikely.*)
    *(.text.startup .text.startup.*)
    *(.text .text.*)
    *(.gnu.linkonce.t.*)
	. = ALIGN(8);
  } > RAM

  .rodata	:
  {
    *(.rdata)
    *(.rodata .rodata.*)
    *(.gnu.linkonce.r.*)
	. = ALIGN(8);
  } > RAM

  .data	:
  {
	*(.data .data.*)
    *(.gnu.linkonce.d.*)
	. = ALIGN(8);
  } > RAM
  .sdata	:
  {
	. = ALIGN(8);
    __global_pointer$ = . + 0x800;
    *(.sdata .sdata.*)
    *(.gnu.linkonce.s.*)
    . = ALIGN(8);
    *(.srodata .srodata.*)
	. = ALIGN(8);
  } > RAM

  .sbss : {
    *(.sbss .sbss.* .gnu.linkonce.sb.*)
    *(.scommon)
	. = ALIGN(8);
  } > RAM
  .bss :
  {
	*(.bss)
	. = ALIGN(8);
  } > RAM

  _end = .;

  .stack :
  {
	_heap_end = .;
	. = . + __stack_size;
	_sp = .;
  } > RAM
}
```

The trace seems to imply that value `0` is read from memory and restored to `ra` thus the function returns not to the normal return location but to program start.

If `rept 7` is replaced by `rept 6` things run perfectly. My interpretation is that in one case the read to `ra` is not actually performed from RAM but from a write buffer, while in the other case a read from RAM is performed and the simulated RAM answers `0`.
 
I'm at a loss—how am I supposed to run programs that write to RAM?

aprnath, 2020-02-18:
Hi,
Are you using the provided simple testbench for this test? If so, then yes there is a known issue that this demo testbench does not support proper store/load persistence.

We are going to be releasing a more capable testbench slave shortly. Stay tuned.

monniaux, 2020-02-19:
Indeed. I just use my own assembly code instead of `hello_world2.s`. Since I could not make head or tail of what was happening, I also fiddled with the link file.

Do I understand it correctly that this demo does not support writing to RAM and reading back from it?

aprnath, 2020-02-19:
> Do I understand it correctly that this demo does not support writing to RAM and reading back from it?

Yes. Sorry. A better demo testbench will be pushed soon.

agrobman, 2020-02-19:
meanwhile you can try to update testbench/ahb_sif.sv

always @ (posedge HCLK or negedge HRESETn) begin
  if (Write **&& Last_HADDR == 32'h0**) begin

removing '&& Last_HADDR ==32'h0'

aprnath, 2020-02-20:
Please try out the latest release and reopen if needed.

closed 2020-02-20 by aprnath