chipsalliance/Cores-VeeR-EH1 #37: Question: how to disable (bypass) data checks in cache?

vit82, 2020-02-20:
I see that CPU often checks for instruction data in icache even when it's disabled by SWERV configuration (icache_enable=0). This affects overall CPU memory access performance. 
Please, suggest if it's possible to bypass icache.

aprnath, 2020-02-20:
The core is designed to have ICACHE , ICCM or both. It is not optimized for the case that Icache and ICCM are **both** not enabled. It will function but not efficiently. In other words , cycles are not taken out when Icache is not enabled.

agrobman, 2020-02-20:
check for RV_ICACHE_ENABLE define in common_defines.vh if this defined the IC logic will present in design. if you want CPU without IC, you need to use -unset icache_enable switch with swerv.config
 (-set icache_enable=0, has no effect) ...

closed 2020-02-26 by vit82