chipsalliance/Cores-VeeR-EH1 #39: the design won't synthesize in Vivado due to syntax errors

monniaux, 2020-02-24:
The Vivado synthetizer does not like the ' in
        rden <= '0;
        wren <= '0;

I don't know anything about Verilog.
The following fixes the syntax issues but I'm unsure whether this actually works :
```
diff --git a/design/dmi/dmi_jtag_to_core_sync.v b/design/dmi/dmi_jtag_to_core_sync.v
index aa4c19f..2d1477c 100644
--- a/design/dmi/dmi_jtag_to_core_sync.v
+++ b/design/dmi/dmi_jtag_to_core_sync.v
@@ -48,8 +48,8 @@ module dmi_jtag_to_core_sync (
 // synchronizers  
 always @ ( posedge clk or negedge rst_n) begin
     if(!rst_n) begin
-        rden <= '0;
-        wren <= '0;
+        rden <= 0;
+        wren <= 0;
     end
     else begin
         rden <= {rden[1:0], rd_en};
```

aprnath, 2020-02-24:
Thanks for pointing this out. The change for PR #29 somehow got dropped in this release.

olofk, 2020-02-24:
That's strange. I just updated [SweRVolf](https://github.com/chipsalliance/Cores-SweRVolf) to use SweRV 1.5 and didn't see any issues with Vivado. Doing standalone Vivado synthesis (e.g. with `fusesoc run --target=synth chipsalliance.org:cores:SweRV_EH1 --part=xc7a100tcsg324-1`) also works fine

aprnath, 2020-02-24:
Re-applied PR #29 via commit cb5a7a141df1151e21221c8b1489864e1b6a7070
Thanks for reporting this.

closed 2020-02-24 by aprnath