chipsalliance/Cores-VeeR-EH1 #44: Problems with 1.5 in SweRV_fpga

danidep02, 2020-02-27:
Hi,
I was using SweRV_fpga (https://github.com/chipsalliance/Cores-SweRV_fpga) with version 1.4 of SweRV with no problems. Everything was working fine.

When I've changed the core to SweRV 1.5, OpenOCD is failing. This is the message that I obtain when running OpenOCD:
   Open On-Chip Debugger 0.10.0+dev-00530-gaf3a034b5 (2020-02-07-14:17)
   Licensed under GNU GPL v2
   For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
   adapter speed: 1500 kHz
   Info : clock speed 1500 kHz
   Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
   Error: Debug Module did not become active. dmcontrol=0x0
   Info : Listening on port 3333 for gdb connections
   Error: Target not examined yet

Do you know which can be the reason?

Thanks,
Dani

aprnath, 2020-02-27:
Hi,
Please note that [release 1.5](https://github.com/chipsalliance/Cores-SweRV/blob/master/release-notes.md) had an addition port added (`dbg_rst_l`) which is a system wide reset for the debug module, independent from the core reset. At the very least this should be connected to the core reset signal. If left unconnected, the debug module's behavior would be unpredictable.

Please check this and let us know if you still have issues.

danidep02, 2020-02-28:
Hi,
I've connected the new port to the core reset and it works :)
Thanks!

aprnath, 2020-02-28:
Cool. Thanks for confirming.

closed 2020-02-28 by aprnath