chipsalliance/Cores-VeeR-EH1 #50: external interrupt handler return problem

vit82, 2020-04-03:
When writing External interrupt handler in assembly language, I follow advice from the datasheet:
[image]

However the CPU does not return on MRET isntruction to the main code. It continues execution of the instructions below MRET. One workaround is to place about 20 NOP instructions before MRET. In that case the CPU executes MRET as expected - it jumps to the address stored in MEPC.

Please, advice whether I miss something.

agrobman, 2020-04-04:
How do you initialize base field in meihap? how do you set up the interrupt vector table? what value do you get in t1  before jr t1? does it match with your interrupt handler start address?
can you single step your handler? can you get trace info from the CPU ( with logic analyzer)?
Can you simulate your code on RTL or ISS and get program trace ?

vit82, 2020-04-06:
At the moment I simulate RTL only. Please, have a look at the waveform with the requested signals.Also attached are program code and its listing file.
[image]
[issue50_listing.txt](https://github.com/chipsalliance/Cores-SweRV/files/4438446/issue50_listing.txt)
[issue50_program.txt](https://github.com/chipsalliance/Cores-SweRV/files/4438447/issue50_program.txt)

Register t3(x28) is used in the program insted t1. Before executing jr t3, t3 gets value 0x100016 - this is address of the desired interrupt handler (UART). So this looks correct.

MEIHAP gets value 2, this corresponds to uart_irq (level sensitive) connected to extintsrc_int[2]. The interrupt handler pointer (uart_handler) is stored in DCCM (offset 0x200000).

vit82, 2020-04-06:
At the waveform PC_experssion jumps to 0x100016(uart_handler) after 0x100074. Next, CPU clears interrupt source UART_IRQ, but it takes quite many cycles. If I don't insert NOPs, MRET occurs early while UART_IRQ is still high. This is probably why MRET isn't executed.

agrobman, 2020-04-06:
You can use tracer code from tb_top.sv to produce CPU trace to debug your simulations.

agrobman, 2020-04-06:
you probably need to poll UART  interrupt pending register and CPU MIP until no pending interrupts detected before exit the interrupt handler ..

aprnath, 2020-04-13:
> At the waveform PC_experssion jumps to 0x100016(uart_handler) after 0x100074. Next, CPU clears interrupt source UART_IRQ, but it takes quite many cycles. If I don't insert NOPs, MRET occurs early while UART_IRQ is still high. This is probably why MRET isn't executed.

You must clear the IRQ source and validate that the IRQ has been cleared before executing MRET. Else the core will come right back into the handler if the input external interrupt pin is still high when the MRET executes.

Please open a new issue if you still have questions.

closed 2020-04-13 by aprnath