chipsalliance/Cores-VeeR-EH1 #51: About OpenOCD extensions

linuxbest, 2020-04-26:
Hi, all,

It seems standard OpenOCD can not access ICCM/DCCM, how to enable it?

There some information mentioned here, Anyone has the detail information?

Thanks.

SweRV_CoreMark_Benchmarking.pdf Page 7:
"
We added abstract commands (based on RISC-V Debug
specification 0.13) to access the ICCM and DCCM
(leverage and extend CREX OpenOCD work)
"

agrobman, 2020-05-14:
A few things need to be added:
1) A way to instruct the tool to use memory abstract commands, depending on address (for PIC DCCM/ICCM address ranges) for data dumps/updates. these accesses must be of word size and aligned. Read/Modify/Write operations should be used for smaller sizes data updates. 
2) the same for SW breakpoints settings in ICCM 
3) add IC invalidation code for SW breakpoint setting with a CPU dmst (0x7C4) SCR via abstract command.

agrobman, 2020-07-16:
See also

https://github.com/chipsalliance/Cores-SweRV/issues/65#issuecomment-657990440

aprnath, 2020-08-17:
Codasip has a patched OpenOCD version for SweRV that has proper support for accessing the internal memories: https://github.com/Codasip/swerv-openocd. If you'd like to try it, please see the README for instructions.

Thanks!

closed 2020-08-17 by aprnath