chipsalliance/Cores-VeeR-EH1 #56: Set RV_FPGA_OPTIMIZE by default?

wsnyder, 2020-05-31:
As described in a recent paper on optimizing SweRV performance on Verilator (see under https://veripool.org/papers) there is a very large performance speedup when RV_FPGA_OPTIMIZE is set.

I would suggest setting this by default in your simulation makefiles, for all three cores.

I would also suggest it is much more likely a given user will be doing FPGA design than ASIC design, and you may want to make it always be the default for all usages and require/document -set=fpga_optimize=0 for ASIC users.

aprnath, 2020-06-03:
Hi Wilson,
 We will make the change as requested at the next update. Some leftover clock gating has been identified by Codasip that did not get cleaned up with RV_FPGA_OPTIMIZE, which will also be fixed in the upcoming release.
Thanks for your continued push towards improving SweRV.

aprnath, 2020-07-06:
RV_FPGA_OPTIMIZE is now the default build option. Please see README for details.

closed 2020-07-06 by aprnath