chipsalliance/Cores-VeeR-EH1 #57: verilator in the make file not working

mikemusasa, 2020-06-11:
Hi ! 
 I have been trying to the  make -f $RV_ROOT/tools/Makefile  
 and the verilator build is failing . 

but when i run the verilator is installed in my  ubuntu system and the commands works se proof:   
mikemusasa@SE-00023761:~/Cores-SweRV$ verilator
Usage:
        verilator --help
        verilator --version
        verilator --cc [options] [source_files.v]... [opt_c_files.cpp/c/cc/a/o/so]
        verilator --sc [options] [source_files.v]... [opt_c_files.cpp/c/cc/a/o/so]
        verilator --lint-only    [source_files.v]...

And the error i am getting from the make file is this : 

%Error: Command Failed /verilator_bin -UASSERT_ON --cc -CFLAGS '-std=c++11' snapshots/default/common_defines.vh /home/mikemusasa/Cores-SweRV/design/include/swerv_types.sv -I/home/mikemusasa/Cores-SweRV/design/include -I/home/mikemusasa/Cores-SweRV/design/lib -Isnapshots/default -Wno-UNOPTFLAT -I/home/mikemusasa/Cores-SweRV/testbench -f /home/mikemusasa/Cores-SweRV/testbench/flist /home/mikemusasa/Cores-SweRV/testbench/tb_top.sv /home/mikemusasa/Cores-SweRV/testbench/ahb_sif.sv --top-module tb_top -exe test_tb_top.cpp --autoflush
/home/mikemusasa/Cores-SweRV/tools/Makefile:85: recipe for target 'verilator-build' failed
make: *** [verilator-build] Error 127 

Any help would be appriciated

aprnath, 2020-06-11:
The "**/**" in "/verilator_bin" in the make error message is suspicious. It should be the full path, or just the command (if verilator is in your $PATH).

mikemusasa, 2020-06-11:
Hi,  
The verilator is in my PATH.   Do you have any idea what could cause this error message ?

aprnath, 2020-06-11:
Is verilator_bin installed in the root dir "/" ? The message you pasted seems to indicate it is trying to execute verilator_bin from /verilator_bin .. which seems an unlikely path to find verilator_bin.

Have you been able to execute Verilator standalone? 
Try copying the failed command fully into your shell and see if it works.

mikemusasa, 2020-06-11:
Hi , Yes i have been able to run the verilator command alone .    I think verilator_bin is in my root . 
when i tried to run the full command , i got a command not found

aprnath, 2020-06-12:
Well, unless you modified the Makefile, which does not specify a path for verilator, only thibg I can think of is that your verilator install is not complete. The Makefile just runs `verilator`
    VERILATOR = verilator  
        : 
    $(VERILATOR) '-UASSERT_ON' --cc -CFLAGS ${CFLAGS} $(defines) $(includes)

aprnath, 2020-06-12:
Sorry.. didn't mean to close it without a resolution.

mikemusasa, 2020-06-16:
No problem, i had a issue with my toolchain i think and ubuntu. Deleted everything and reinstalled, it worked

closed 2020-06-16 by mikemusasa