chipsalliance/Cores-VeeR-EH1 #59: problem with the linker and undefined reference  errors ?

mikemusasa, 2020-06-22:
HI ! 
Is there a problem with the linker when running a c code ? 
My c code includes all the necessary librarys and compiles separatly without a problem   but i try to run it using the core , I keep getting undefine reference error for a lot of standard c funtions  example printf ? anyone else had this error ?

agrobman, 2020-06-28:
You have to review your code and find what is missing (what linker complains about)

mikemusasa, 2020-06-28:
I review my code, now whe i try to run it , i am getting some undefined reference errors to simple c funtions Like printf , fread, fopen even the headers to includes these librarys are there.   I saw in the c mark code that they had to write their own printf type of functions .  those the linker have a problem finding their definitions ?

agrobman, 2020-06-28:
The "simple C functions" are not that simple - there is a bunch of code to be executed by the CPU and computer HW/OS/BIOS.
 
You have to link all used libraries - Swerv Makefile doesn't do this. 

However, non of the standard I/O library functions will work on provided TB.

Swerv TB can't execute OS functions as fopen or printif, which require some OS and HW.
It does not have screen/display controller or any hard drive to support file system. 
I'm not sure that "-unknown-" tool chain even have all these I/O libraries, which are OS dependent...

You have to provide your own implementation for any I/O functions your program is using. 

cmark code is an example of simple terminal emulation with printif function, based on putc function, which in the swerv TB
just writes to a specific address.  The TB  monitors writes to this address on LSU bus to print on screen and a log file.

closed 2021-01-22 by aprnath