chipsalliance/Cores-VeeR-EH1 #60: Add GitHub repository topics

dominiksalvet, 2020-06-22:
Hello SweRV community! :wave:

It would be great to add some relevant GitHub topics to this repository. It may bring more people and help with promoting this core as well as related open-source stuff. What do you think? :thinking:

aprnath, 2020-06-22:
Hi Dominik,
What kind of topics did you have in mind? Some examples/suggestions ?

dominiksalvet, 2020-06-22:
Yeah, e.g., [`processor`](https://github.com/topics/processor), [`risc`](https://github.com/topics/risc), [`riscv`](https://github.com/topics/riscv)/[`risc-v`](https://github.com/topics/risc-v), [`swerv`](https://github.com/topics/swerv) (!), [`open-source`](https://github.com/topics/open-source), [`hardware`](https://github.com/topics/hardware), [`pipeline`](https://github.com/topics/pipeline), [`32-bit`](https://github.com/topics/32-bit), [`wd`](https://github.com/topics/wd), [`superscalar`](https://github.com/topics/superscalar), [`axi`](https://github.com/topics/axi)/[`ahb`](https://github.com/topics/ahb), [`dma`](https://github.com/topics/dma), [`seh1`](https://github.com/topics/seh1)/[`eh1`](https://github.com/topics/eh1), [`pic`](https://github.com/topics/pic), [`systemverilog`](https://github.com/topics/systemverilog)/[`verilog`](https://github.com/topics/verilog), [`hdl`](https://github.com/topics/hdl), [`rtl`](https://github.com/topics/rtl), [`fusesoc`](https://github.com/topics/fusesoc), [`verilator`](https://github.com/topics/verilator), [`fpga`](https://github.com/topics/fpga), [`asic`](https://github.com/topics/asic), ... that should be enough. Feel free to pick the most suitable for you. :smile: :rocket:

aprnath, 2020-06-22:
Great idea. Thank you for the suggestion!

closed 2020-06-23 by dominiksalvet