chipsalliance/Cores-VeeR-EH1 #63: $readmem file address beyond bounds of array

cebaut, 2020-07-03:
Hi,

 I try to simule a small file.c that it uses standard libraries (stdio).
For example to use printf from stdio: 
```
#include <stdio.h>
#include "defines.h"
#define ITERATIONS 1
extern int STACK;
int main();

int mult() {
        int a=1000,b=3;
        return a*b;
}

/*********************MAIN************************/

int  main() {
        int result;
        result = mult();
        printf("printf" );
	
return 0;

}
```

I needed to change the linker script like this :
```
_OUTPUT_ARCH( "riscv" )
ENTRY(_start)

SECTIONS
{
 _start = .;
    . = 0;
  .text   : { *(.text*) }
 _end = .;
  .data  :  ALIGN(0x800) { *(.*data) *(.rodata*) STACK = ALIGN(16) + 0x8000; }
}
```_

and now I have a problem I think with verilator when I try t0 simule : 
`VerilatorTB: Start of sim`
`%Error: program.hex:4097: $readmem file address beyond bounds of array`

I would like to know how configure more memory to execute my test? 

Thanks,

cebaut :)

agrobman, 2020-07-06:
testbench/ahb_sif.sv memory slave has following parameter, defining size of the external slaves in double words. 
This limits code and  data size to 64KB each.

"
parameter MEM_SIZE_DW = 81920;
"

However, with memories size increase,  you may need to change the tools/Makefile too on how to generate the hex files.

Also you probably will need to provide your own implementation on printf/putc functions to get something useful from usage of  printf in swerv demo TB, as standard printf will do nothing there.

cebaut, 2020-07-09:
Hi, thank you for your answer.

I put parameter MEM_SIZE_DW = 16384 to have this limits code and data size to 128KB each.
I launch `make -f $RV_ROOT/tools/Makefile verilator TEST=tst `
but it's always the same problem.
I don't understand what I have to change in tools/Makefile to achieve.
I would like to have a printf from the library standard on my simulation.

 I don't know what is exactely a TB ?

Thaks,

cebaut :)

agrobman, 2020-07-09:
1.  TB – testbench, Verilog environment where the CPU RTL is installed; ( tb_top.sv in Swerv case)
  2.  With increased memory size you need to place data at address 0x20000  in link.ld , changing statement “.= 0x10000;” to “.=0x20000;”
  3.  To link STDIO library – you’ll need to change linker command to add the library.
  4.  Again, standard I/O functions won’t work , you’ll probably can replace “putc” library function by something like:

int putc( char ch){
                *(volatile int *)(0xd0580000) = ch;
             Return ch;
}

TB snoops this address and prints data, as characters to the screen during simulation

--

cebaut, 2020-07-09:
Thanks !
I try that....

cebaut, 2020-07-09:
Hi,
always same message
`%Error: program.hex:4097: $readmem file address beyond bounds of array`
:(
My linker script:

OUTPUT_ARCH( "riscv" )
ENTRY(_start)

SECTIONS
{
 _start = .;
    . = 0;
  .text   : { *(.text*) }
 _end = 0x20000;
  .data  :  ALIGN(0x800) { *(.*data) *(.rodata*) STACK = ALIGN(16) + 0x8000; }
}
```
4.  Again, standard I/O functions won’t work , you’ll probably can replace “putc” library function by something like:

int putc( char ch){
                *(volatile int *)(0xd0580000) = ch;
             Return ch;
}
```
=> I don't do that for the moment !

any idea ?

agrobman, 2020-07-09:
Probably your .text is longer than 128K.

Linker may produce  a map file to see the memory allocations.

Also you can use *_objdump utility to see content of  the executable

-H option gives you output sections info – start address , size , etc
-D allow to disassemble executable.

Also the linker file should be:

OUTPUT_ARCH( "riscv" )
ENTRY(_start)

SECTIONS
{
    . = 0;
  .text   : { *(.text*) }
_end = .;
  . = 0x20000;
  .data  :  ALIGN(0x800) { *(.*data) *(.rodata*) STACK = ALIGN(16) + 0x8000; }
}

For your size of the memories
--

closed 2021-01-22 by aprnath