chipsalliance/Cores-VeeR-EH1 #66: Determine size of i-cache

wronkiew-ghs, 2020-07-12:
PRM 8.5.1 says the dicawics index field is right-aligned with unused bits 0. If the unused bits read as zero, the debugger can set them and read back the size of the i-cache. We configured EH1 with a 16-KB icache. Setting the top bits of index reads back 1's. Is this correct operation? If so, is there another way to determine the number of cache lines implemented?

tmw-wdc, 2020-07-14:
The I-cache diagnostic access registers don't provide the capability to determine the I-cache size.  The index field of the dicawics register does not have WARL behavior.  However, this contradicts the statement in Section 8.5 that, unless otherwise noted, all read/write fields are WARL.  We will add a note to the index field description clarifying this.
Thank you for bringing this to our attention!

wronkiew-ghs, 2020-07-15:
Thank you for the explanation.

closed 2020-07-15 by wronkiew-ghs