chipsalliance/Cores-VeeR-EH1 #69: Cores-SweRV development planning

zhuzhizhan, 2020-08-20:
Hi,

I am a software engineer at Rockchip. I want to evaluate this core. Is there a long-term support plan for this core？

zhuzhizhan, 2020-08-24:
I'm evaluating this core and we will use it to design our SOC if it work well. So if WDC has a long term support for this core, this will help us encounter less trouble.
Some more questions:
1.Where is the core's software sdk?
2.Can I use a normal GCC to compile the code with this superscalar core. Or can WDC provide the toolchain？

JanMatCodasip, 2020-08-24:
>  1.Where is the core's software sdk?

Please take a look at: https://github.com/westerndigitalcorporation/riscv-fw-infrastructure

>  2.Can I use a normal GCC to compile the code with this superscalar core. Or can WDC provide the toolchain？

Yes, "normal" bare-metal RISC-V GCC toolchain can be used.

aprnath, 2020-08-24:
WDC will provide HW bug fixes as discovered internally or reported by the community. I would suggest you consult the good folks at Codasip, who are proving commercial support for this core if you need commercial grade support for your projects.

zhuzhizhan, 2020-08-25:
> WDC will provide HW bug fixes as discovered internally or reported by the community. I would suggest you consult the good folks at Codasip, who are providing commercial support for this core if you need commercial grade support for your projects.

Got it, thanks.  So is the core used by Codasip based from Cores-SweRV? And do they have made some extensions for the core or just provide some external IP?

Codasip, 2020-08-25:
We are providing SweRV Core Support Package containing proven scripts, SDK , examples and documentation. Coremark, dryhstone and mbench are included with "how to" description as well. There is a "SweRV Core Support Package Pro, for those who wants to have direct support with defined response time. We are also doing some extensions for the core, however this is typically treated as dedicated project - additional to the support.

zhuzhizhan, 2020-08-25:
> We are providing SweRV Core Support Package containing proven scripts, SDK , examples and documentation. Coremark, dryhstone and mbench are included with "how to" description as well. There is a [...]

Got it. So do Codasip have a long term support for this core, include updating risc-v new features and so on?

Codasip, 2020-08-25:
Exactly. If you have a concrete project and concrete features which requires "pro" support do not hesitate to contact us. For open questions and discussions you are right here.

closed 2020-08-25 by aprnath