chipsalliance/Cores-VeeR-EH1 #70: typo in swerv.config

sobuch, 2020-08-24:
In https://github.com/chipsalliance/Cores-SweRV/blob/master/configs/swerv.config#L1914
```
 $jh{memmap}{cosnoleio} = $config{memmap}{serialio} if exists $config{memmap}{serialio};
```
I suppose it should be: cosnoleio -> consoleio

aprnath, 2020-08-25:
Thanks for reporting. Note that this is only used by the testbench, and not the design itself. 
Will be fixed in next release.

sobuch, 2020-09-21:
I see this changed in the new version, however it still doesn't work for me.

Currently, there is the following at https://github.com/chipsalliance/Cores-SweRV/blob/master/configs/swerv.config#L1962:
```
    foreach my $tag (qw ( size page_size serialio )) {
        $jh{memmap}{tag} = $config{memmap}{ta} if exists $config{memmap}{tag};
    }
```
Could it be more like this?
```
    $config{memmap}{consoleio} = $config{memmap}{serialio} if exists $config{memmap}{serialio};

    foreach my $tag (qw ( size page_size serialio consoleio)) {
        $jh{memmap}{$tag} = $config{memmap}{$tag} if exists $config{memmap}{$tag};
    }
```

aprnath, 2020-09-22:
Indeed. Good catch Sam.

sobuch, 2020-09-23:
I can confirm its fine now, thus closing this issue. Thank you for looking into this.

closed 2020-09-23 by sobuch