chipsalliance/Cores-VeeR-EH1 #71: dccm initialization

S-Nomii, 2020-09-01:
Hi, I was trying to understand how dccm works, but it keep reporting assertion fail for output data port of dccm is always x. I was trying to use '.word' in my assembly file, and change the link.ld file to make the address fit the dccm region, but the dccm is never loaded with those word i write. I am curious about when and how dccm load the word from my data.hex file?

agrobman, 2020-09-01:
try to understand hello_world_dccm.ld/cmark_dccm.ld - examples how to create executables with DCCM preload for simulations. In real life, CPU has to write DCCM with aligned word size stores before use in program. CPU Boot code is the place for DCCM initialization.

S-Nomii, 2020-09-02:
> try to understand hello_world_dccm.ld/cmark_dccm.ld - examples how to create executables with DCCM preload for simulations. In real life, CPU has to write DCCM with aligned word size stores before use in program. CPU Boot code is the place for DCCM initialization.

Thanks for explaining that, I will check it later, and yesterday I noticed another problem. The store instruction get permission to write back to dccm at dc5 stage, but if I launch an instruction to load data from the same memory space right after that store instruction, the load instruction tried to read data from dccm at dc2 stage and find x in its output port. Then it will raise an assertion failure and I can't find the logic that stop the load instruction from launching. 

example code

li x1, 0xf004000
li x2, 0xffffffff
sw x2, 0x0(x1)
lw x3, 0x0(x1)

S-Nomii, 2020-09-02:
Thanks for explaining that, I thought it is the hardware's job to link data between dccm with memory. I noticed another problem yesterday. The store instruction write data back to dccm at dc5 stage, but if I launch a load instruction accessing the same memory space, it tried to read data from dccm at dc2 stage. Since the data has not been written yet, it will read x in the output port and raise several assertion failure.  So I'll be very grateful if you can help me understand how eh1 deal with these sort of reference (if it does) , so I can eliminate those assertion failure.

sample code 
li x2, 0xf004000
li x3, 0xffffffff
sw x3, 0x0(x2)
lw x4, 0x0(x2)

S-Nomii, 2020-09-02:
> try to understand hello_world_dccm.ld/cmark_dccm.ld - examples how to create executables with DCCM preload for simulations. In real life, CPU has to write DCCM with aligned word size stores before use in program. CPU Boot code is the place for DCCM initialization.

Thanks for explaining that, I thought it is the hardware's job to link data between dccm and memory. I noticed another problem yesterday. The store instruction write data back to dccm at dc5 stage, but if I launch a load instruction accessing the same memory space, it tried to read data from dccm at dc2 stage. Since the data has not been written yet, it will read x in the output port and raise several assertion failures. So I'll be very grateful if you can help me understand how eh1 deal with these sort of references or data hazard (if it does).

sample code
li x2, 0xf004000
li x3, 0xffffffff
sw x3, 0x0(x2)
lw x4, 0x0(x2)

aprnath, 2020-09-02:
Comments in issue #36 may clarify this for you..

closed 2021-01-22 by aprnath