chipsalliance/Cores-VeeR-EH1 #72: The CSR mepc cann't save correct mret address after Continuous NMI happened

Richard2088, 2020-09-04:
Hi Sir:
    When the Continuous NMI happened,  mepc cann't save the return address correctly.
   
   And if we Triggered NMI every more then ten milliseconds, everything is ok!

program 1： 
20000816:	508010ef          	jal	ra,20001d1e <SMPU_Add_Region>
2000081a:	4585                	li	a1,1
2000081c:	4501                	li	a0,0
2000081e:	4b2010ef          	jal	ra,20001cd0 <SMPU_Enable>
20000822:	20020737          	lui	a4,0x20020
**20000826:	431c                	lw	a5,0(a4) //This LW instruction triggers NMI, and the value stored in MEPC is 20000826**
20000828:	e0000537          	lui	a0,0xe0000
2000082c:	22c50513          	addi	a0,a0,556 # e000022c <__freertos_irq_stack_top+0xfffc022c>
20000830:	0887e793          	ori	a5,a5,136
20000834:	c31c                	sw	a5,0(a4)
20000836:	496020ef          	jal	ra,20002ccc <puts>
2000083a:	4501                	li	a0,0
2000083c:	6ed010ef          	jal	ra,20002728 <exit>

program 2：
20000816:	518010ef          	jal	ra,20001d2e <SMPU_Add_Region>
2000081a:	4585                	li	a1,1
2000081c:	4501                	li	a0,0
2000081e:	4c2010ef          	jal	ra,20001ce0 <SMPU_Enable>
20000822:	200207b7          	lui	a5,0x20020
**20000826:	438c                	lw	a1,0(a5)//This LW instruction triggers NMI, and the value stored in MEPC is 20002b00**
20000828:	e0000537          	lui	a0,0xe0000
2000082c:	22c50513          	addi	a0,a0,556 # e000022c <__freertos_irq_stack_top+0xfffc022c>
20000830:	2d0020ef          	jal	ra,20002b00 <iprintf>
20000834:	e0000537          	lui	a0,0xe0000
20000838:	24450513          	addi	a0,a0,580 # e0000244 <__freertos_irq_stack_top+0xfffc0244>
2000083c:	4a0020ef          	jal	ra,20002cdc <puts>
20000840:	4501                	li	a0,0

program 3：
20000816:	508010ef          	jal	ra,20001d1e <SMPU_Add_Region>
2000081a:	4585                	li	a1,1
2000081c:	4501                	li	a0,0
2000081e:	4b2010ef          	jal	ra,20001cd0 <SMPU_Enable>
20000822:	200207b7          	lui	a5,0x20020
20000826:	08800713          	li	a4,136
2000082a:	e0000537          	lui	a0,0xe0000
**2000082e:	c398                	sw	a4,0(a5) //This SW instruction triggers NMI, and the value stored in MEPC is 20002b6c**
20000830:	22c50513          	addi	a0,a0,556 # e000022c <__freertos_irq_stack_top+0xfffc022c>
20000834:	498020ef          	jal	ra,20002ccc <puts>
20000838:	4501                	li	a0,0
2000083a:	6ef010ef          	jal	ra,20002728 <exit>

Looking forward to your reply
Thanks!
Best wish.

aprnath, 2020-09-04:
NMIs are  nonmaskable and not deferred (and usually fatal), so if an NMI  occurs while you are still handling a prior exception/interrupt, there is no built-in protection available to the processor states to allow recovery, since the hardware architecture does not have provision hardware for nested traps or a trap stack.

If you need to be able to handle nested NMIs, you could try to save off MEPC into a stack structure as the first thing in your handler.. but even that is not a guarantee - since you could really get them back to back in a pathological case.

agrobman, 2020-09-04:
BTW, why do you consider that 1st case is correct and 2,3 are not?

Richard2088, 2020-09-07:
> NMIs are nonmaskable and not deferred (and usually fatal), so if an NMI occurs while you are still handling a prior exception/interrupt, there is no built-in protection available to the processor states to allow recovery, since the hardware architecture does not provision hardware for nested traps or a trap stack.
> 
> If you need to be able to handle nested NMIs, you could try to save off MEPC into a stack structure as the first thing in your handler.. but even that is not a guarantee - since you could really get them back to back in a pathological case.

Thank you for your reply！
These NMIs are not generated in exceptions or irq, but in normal procedures.
   It is not the problem of saving empc with the stack, the entry function mepc in nmi_entry is wrong.

Richard2088, 2020-09-07:
> BTW, why do you consider that 1st case is correct and 2,3 are not?

Thank you for your reply！
According to the description of mecp in the document: mepc will save the next instruction that occurs in NMI, so I think it is an error if it does not return normally.

agrobman, 2020-09-07:
how do you know instruction when NMI did occur? How is the NMI generated, "triggered" for your cases? 

The Swerv CPU is pipelined machine.  Fetches, instruction commits and Load/Store bus transactions happen in different times. Loads are "non-blocking" and may write target register much later than the loads are "committed" or "executed" from CPU pipe point of view. Any bus errors, caused by external Loads/Store instructions are considered as "asynchronous" to the instruction flow and can be "attached" to any instruction, followed the Load/store instruction with bus error. These NMI exceptions are not "precise" in sense the MEPS does not points on instruction, really caused the exception. For the same code snippet, MEPC can be different, depending on execution and bus timings, even if the same Load/Store instruction is causing the bus error.  

The stores are  handled  with a store buffer and considered by the CPU "executed", when issued to the store buffer, but actually create write transactions on the LSU bus later in time. The store bus errors are also "asynchronous" and not precise.

Only load/store errors to DCCM/PIC or external memory with side-effect are handled precisely ...

SW can't use mepc to point on  instruction, caused NMI exception, related to the bus error...

Richard2088, 2020-09-07:
> how do you know instruction when NMI did occur? How is the NMI generated, "triggered" for your cases?
>
> The Swerv CPU is pipelined machine. Fetches, instruction commits and Load/Store bus [...]

Thank you for your reply！
     NMI does not happen inside the core, but outside.  In a normal program, when I use a SW/LW instruction to access the address protected by the MPU, and trigger NMI, it enter nmi_entry and cannot jump to the correct execution address after 'mret'.  We checked the simulation waveform, and did not see the bus error or other errors, but the value saved by MEPC was incorrect, which caused the program jump error after nmi_entry 'mret'.

Thanks!

agrobman, 2020-09-07:
What mcause value do yo get in the exception handler?
Based on my knowledge, MPU violation should cause load access fault (mcause=5) precise exception  and not NMI.

Richard2088, 2020-09-08:
> What mcause value do yo get in the exception handler?
> Based on my knowledge, MPU violation should cause load access fault (mcause=5) precise exception and not NMI.

Thank you for your reply！ Brother.
mcause is not just the cause of the NMI error. When the NMI described above occurs, mcause is 0.
I only use MPU to trigger the NMI problem, and the software can only use this to trigger NMI. The same phenomenon is true if the verification environment sends other NMIs.

agrobman, 2020-09-08:
What MPU are you referring to ? Is this core external unit or SweRV built-in address protection, set up by swerv.config parameters?
If your testbench asserts NMI request pin, you can't expect any specific value in MEPC when NMI exception is taken, unless you tune NMI pin assertion timing specifically for your code  snippet  execution .

BTW, what do you call "other NMIs"?  is this NMI pin assertion or something else? CPU also takes NMI, when gets data bus errors..

agrobman, 2020-09-08:
NMI works similar way as external or timer  interrupt - generally you can't predict at what address your program will be interrupted ...

Richard2088, 2020-09-09:
> What MPU are you referring to ? Is this core external unit or SweRV built-in address protection, set up by swerv.config parameters?
> If your testbench asserts NMI request pin, you can't expect any [...]

MPU：Memory Protect Unit， Between the bus and flash, access to flash through the bus requires the judgment of MPU。
other NMIs：laser NMI etc.

> What MPU are you referring to ? Is this core external unit or SweRV built-in address protection, set up by swerv.config parameters?
> If your testbench asserts NMI request pin, you can't expect any [...]

normal program 1 .dis：
20000816: 508010ef jal ra,20001d1e <SMPU_Add_Region>
2000081a: 4585 li a1,1
2000081c: 4501 li a0,0
2000081e: 4b2010ef jal ra,20001cd0 <SMPU_Enable>
20000822: 20020737 lui a4,0x20020
20000826: 431c lw a5,0(a4) //This LW instruction triggers NMI, and the value stored in MEPC is 20000826
20000828: e0000537 lui a0,0xe0000
2000082c: 22c50513 addi a0,a0,556 # e000022c <__freertos_irq_stack_top+0xfffc022c>
20000830: 0887e793 ori a5,a5,136
20000834: c31c sw a5,0(a4)
20000836: 496020ef jal ra,20002ccc
2000083a: 4501 li a0,0
2000083c: 6ed010ef jal ra,20002728

20000826: trigger NMI， Enter nmi_entry function, return with 'mret'
The program jumps to 20000826 again，Stuck in an endless loop。

And program 2, program 3 as show above,  MEPC The value saved by mepc is random, there is no rule

agrobman, 2020-09-09:
All you see is expected. Depending on execution timing you can get NMI on L/S instruction, which triggers NMI pin assertion or other instructions after. If NMI falls on your triggering L/S instruction you'll have no forward progress. BTW, what exactly you are trying accomplish with this exercise? and you are right - the MEPC value at NMI exception can't be predicted. I wrote already that NMIs are not precise and not recoverable. Once you got NMI, only reliable action you can take is to record the event in memory and reset the system.

Richard2088, 2020-09-09:
> or other instructions after. If NMI falls on your triggering L/S instruction you'll have no forward progress. BTW, what exactly you a

Thank you for your reply！SIR。
 I am practicing the difference in architecture and function between ARM core and RISCV core. I found that contex-m3 can be recovered from NMIs. I can design the threshold of the number of nmi occurrences and how to decide whether to reset system.  However, the riscv core cannot be recovered from the NMI.
In this RISCV CORE situation, the system can only be reset after an NMI occurs.
Thanks！

Richard2088, 2020-09-09:
> All you see is expected. Depending on execution timing you can get NMI on L/S instruction, which triggers NMI pin assertion or other instructions after. If NMI falls on your triggering L/S [...]

> All you see is expected. Depending on execution timing you can get NMI on L/S instruction, which triggers NMI pin assertion or other instructions after. If NMI falls on your triggering L/S [...]

Thank you for your reply！SIR。
I am practicing the difference in architecture and function between ARM core and RISCV core. I found that contex-m3 can be recovered from NMIs. I can design the threshold of the number of nmi occurrences and how to decide whether to reset system. However, the riscv core cannot be recovered from the NMI.
In this RISCV CORE situation, the system can only be reset after an NMI occurs.
Thanks！

Richard2088, 2020-09-09:
> All you see is expected. Depending on execution timing you can get NMI on L/S instruction, which triggers NMI pin assertion or other instructions after. If NMI falls on your triggering L/S [...]

Thank you for your reply！SIR。
I am practicing the difference in architecture and function between ARM core and RISCV core. I found that contex-m3 can be recovered from NMIs. I can design the threshold of the number of nmi occurrences and how to decide whether to reset system. However, the riscv core cannot be recovered from the NMI.
In this RISCV CORE situation, the system can only be reset after an NMI occurs.
Thanks！

agrobman, 2020-09-09:
In case you want to notify CPU about an access violation, the better way is to drive rresp[0] = 1. The cpu will set data error address register (mdseac) to the faulty access address . CPU will still get  NMI exception, but with more info about the problem.

Richard2088, 2020-09-10:
> In case you want to notify CPU about an access violation, the better way is to drive rresp[0] = 1. The cpu will set data error address register (mdseac) to the faulty access address . CPU will still get NMI exception, but with more info about the problem.

Sir, I just want to achieve this effect.
   The system is reset after 100 NMI visits. If it is less than 100 times, it will continue to run.  Can you advise how to do it?  Thank you

agrobman, 2020-09-10:
there is no reliable way to achieve your goal.

Richard2088, 2020-09-11:
> there is no reliable way to achieve your goal.

If it is not possible to resume operation from NMIs, is this a design flaw in the core?  There are still bugs ?

aprnath, 2020-09-11:
The behavior is as intended. NMIs are not meant to replace regular/periodic/maskable interrupts - which is really what you should be using. The expected use for NMIs are for near fatal hardware errors. It is better/recommended that the system go into some sort of software-initiated-reset or scrub after logging the NMI. And if you get an NMI storm then reset if your only option for any forward progress.

While you **may** recover from a single (or multiple sparse) NMI, they cannot be masked or nested, and therefore you cannot guarantee that a 2nd NMI will not occur while you are "handling" the first. The 2nd will overwrite the MEPC from the 1st. Which is why, in your original case, if you spread them out enough it will work.. but if you get NMIs close together, you will never be able to recover.

Richard2088, 2020-09-14:
> The behavior is as intended. NMIs are not meant to replace regular/periodic/maskable interrupts - which is really what you should be using. The expected use for NMIs are for near fatal hardware errors. It is better/recommended that the system go into some sort of software-initiated-reset or scrub after logging the NMI. And if you get an NMI storm then reset is your only option for any forward progress.
> 
> While you **may** recover from a single (or multiple sparse) NMI, they cannot be masked or nested, and therefore you cannot guarantee that a 2nd NMI will not occur while you are "handling" the first. The 2nd will overwrite the MEPC from the 1st. Which is why, in your original case, if you spread them out enough it will work.. but if you get NMIs close together, you will never be able to recover.

Sir: this is not a question of NMI nesting.
   We have the NMI interrupt source enable bits, and only the bit enbale can trigger the corresponding NMI.
    nmi_entry function: 
                  disable NMI source bits 
                  push stack
                  processing nmi interrupt (only print interrupt source code)
                 pop stack
                 enable NMI source bit 

We have done almost all the experiments, and our conclusion is.
   After the occurrence of NMI, the program cannot continue to execute by returning from MEPC.
   But Contex-M is ok

Richard2088, 2020-09-15:
> The behavior is as intended. NMIs are not meant to replace regular/periodic/maskable interrupts - which is really what you should be using. The expected use for NMIs are for near fatal hardware [...]

lw a0, 0(a1)  //can trigger NMI
lw a2, 0(a3)  //can trigger NMI
When two instructions that can trigger NMI are put together, mepc will not be stored corrected and 'mret' cannot be returned.
   Is the problem caused by CORE dual launch?

agrobman, 2020-09-17:
Not, and  again I would recommend to read RISCV spec/CPU PRM,  before making any assumptions about design correctness. NMI is treated by the RISCV spec and design as un-recoverable event, and CPU can't return to normal execution after it. CPU can provide some hints about the NMI reason, but not always precisely. SW needs to threat these hints with caution . For example, for LSU bus errors, CPU records only first faulty address, and it's not necessary from the first L/S instruction in flow in your example ...

Richard2088, 2020-09-23:
> Not, and again I would recommend to read RISCV spec/CPU PRM, before making any assumptions about design correctness. NMI is treated by the RISCV spec and design as un-recoverable event, and CPU [...]

AS "riscv-privileged-20190608-1.pdf" saied

3.1.15 Machine Exception Program Counter (mepc)
mepc is an MXLEN-bit read/write register formatted as shown in Figure 3.21. The low bit of mepc
(mepc[0]) is always zero. On implementations that support only IALIGN=32, the two low bits
(mepc[1:0]) are always zero.
If an implementation allows IALIGN to be either 16 or 32 (by changing CSR misa, for example),
then, whenever IALIGN=32, bit mepc[1] is masked on reads so that it appears to be 0. This
masking occurs also for the implicit read by the MRET instruction. Though masked, mepc[1]
remains writable when IALIGN=32.
mepc is a WARL register that must be able to hold all valid physical and virtual addresses. It
need not be capable of holding all possible invalid addresses. Implementations may convert some
invalid address patterns into other invalid addresses prior to writing them to mepc.
When a trap is taken into M-mode, mepc is written with the virtual address of the instruction
that was interrupted or that encountered the exception. Otherwise, mepc is never written by the
implementation, though it may be explicitly written by software.

The actual situation now is that mepc cannot guarantee when a continuous NMI interrupt occurs.

closed 2021-01-22 by aprnath