chipsalliance/Cores-VeeR-EH1 #73: pic

S-Nomii, 2020-09-11:
hi, i was using eh1 to study memory access in cpu, and i have trouble understanding the pic module. Since pic module is suppose to handle interruptions, what does it have to do with load and store instruction, and why check if address is in the pic region. I tried to load data from 0xf00c0000, the data never write back to register file.

agrobman, 2020-09-11:
Read PIC chapter of SweRV PRM - Some of the PIC control/status registers are implemented as memory mapped, i.e. accessible with CPU Load/Store instructions and the PIC registers are included in SweRV internal memory region.

S-Nomii, 2020-09-12:
> Read PIC chapter of SweRV PRM - Some of the PIC control/status registers are implemented as memory mapped, i.e. accessible with CPU Load/Store instructions and the PIC registers are included in SweRV internal memory region.

Thanks for explaining its usage to me, but I haven't study interruptions part yet. Can I currently consider that PIC address region is only available under interruption circumstances and not available for normal load/store instruction.

aprnath, 2020-09-13:
The PIC address space is for programming and querying the PIC control registers. It should not be treated as "memory" for load/stores.

S-Nomii, 2020-09-13:
> The PIC address space is for programming and querying the PIC control registers. It should not be treated as "memory" for load/stores.

thanks that is very helpful.

olofk, 2020-09-13:
I would like to clarify a thing. Because the PIC registers are internal to the CPU, you can read and write normally with load and store operations but you won't see any accesses to these addresses on the external buses. It also means you can't access them with a debugger like other memory. You need to use "abstract memory commands" for that.

closed 2020-09-13 by aprnath