chipsalliance/Cores-VeeR-EH1 #74: mcontrol chain bit not WARL on triggers 1 & 3

wronkiew-ghs, 2020-09-13:
Setting a store address & value breakpoint using chained triggers succeeds in setting and reading back the mcontrol registers if triggers 1 & 2 are used, but the breakpoint is hit on instructions which do not match the second trigger. These are the values read from mcontrol & tdata2 on triggers 1 & 2:

t1 mcontrol 0x2be018c2
t1 tdata2 0xf0040005
t2 mcontrol 0x2be81042
t2 tdata2 0xaaaaaaaa

In my test program the trigger fired on writes to 0xf0040004 which had values other than 0xaaaaaaaa. If I used triggers 0 & 1 instead the data HWBP worked correctly. I also found the chain bit could be set and read back on trigger 3.

The EH1 PRM section on mcontrol notes that attempts to set chain on triggers 1 & 3 are ignored because chaining is only supported on pairs 0/1 and 2/3. RISC-V debug 0.13.2 specifies all tdata registers are WARL so the debugger can determine which features are supported.

This case is likely to come up if a store addr & value HWBP is set after a simple exec HWBP.

tmw-wdc, 2020-09-15:
Thank you for reporting this issue.  We are looking into it and will report back.

tmw-wdc, 2020-10-01:
After some internal discussions, we have decided to fix the issue you have reported.  In the next release of the SweRV EH1 core, the _chain_ bit of MCONTROL will have WARL behavior (i.e., the chain bit value will always be '0' for reads of MCONTROL for triggers 1 and 3).
Thank you again for reporting this issue.  Much appreciated!

wronkiew-ghs, 2020-10-01:
Thank you very much for looking into this.

closed 2020-10-01 by tmw-wdc