chipsalliance/Cores-VeeR-EH1 #79: question about adding custom instructions

feiger313, 2020-11-02:
hi, 
l saw some infomation  in README.md .:"If adding/removing instructions, espresso must be installed (used by tools/coredecode)".
Can this increase the custom instructions ，and how to use this script .

Codasip, 2020-11-23:
If you want to "increase the custom instructions" of SweRV core you have to describe them in espresso input format and update existing decoder logic in SweRV core. Assuming you have available all resources your custom instructions need.

Espresso or better said espresso-logic is used to minimize boolean functions.  It is used to generate decoder logic from your customized instruction code input. Input format and the way how to use espresso is described in its man pages. SweRV Cores Support Package ( https://github.com/chipsalliance/Cores-SweRV-Support-Package ) contains already installed espresso integrated with SweRV EH1 core and tools. There is also included description of all Espresso functions.

feiger313, 2020-11-27:
thank you for your help

closed 2020-11-27 by feiger313