chipsalliance/Cores-VeeR-EH1 #81: Macro definitions not being found

rlb1116, 2020-11-23:
I am trying to use this SweRV core in a Quartus project, but I am unable to compile due to definitions of text macros (RV_XXX_YYY) used by design files not being found by the compiler. This results in syntax errors on the lines that try to use them. 

I found most of these have been defined in the common_defines.vh file, which gets created as part of a "snapshot" configuration via the Make flow. However, even though I have added this file to my Quartus project, it does not seem to get recognized, and I noticed that none of the design files explicitly "include" this header file. 

I imagine I could get past this by explicitly including common_defines.vh in every design file that uses one of its macros; however, in addition to the tediousness of this, it seems like that would have already been included if it were truly necessary. Is there a standard, or expected, compilation flow which naturally finds these definitions, and therefore would make explicit inclusion of the header file redundant? If so, what am I missing?

I did notice in the filelist design/flist.questa (of which I included all available files listed into my Quartus project), the file design/include/def.sv does not exist on Github, nor is it created by Make. I am not sure if this is important, or has intentionally been discontinued, but it seemed like there may have been a chance that this could be the "missing link" between the macro definitions and the rest of the design files. 

Any help would be appreciated!

agrobman, 2020-11-23:
the `RV_* defines are created by swerv.config script in snapshots/<target>/common_defines.vh file, which need to be included in list of files for compilation before RTL files one time, something like:

snapshorts/default/common_defines.vh
-f $RV_ROOT/design/flist

plus all needed +incdir +libext and etc ..

flist.questa is outdated, I think. Consult with tools/Makefile how to compile swerv with Questa.
You may need to use full path names to the files ..

Notice that Makefile compiles also DemoTB files you may not need in your project ..

olofk, 2020-11-24:
Hi,

If you use FuseSoC, this will be done automatically for you. Unfortunately it seems like SweRV doesn't work with Quartus (at least not the on the no-cost version) due to some syntax problems. I just tried to do a quick test synthesis of swerv with quartus for a random device using

    fusesoc run --tool=quartus --target=synth chipsalliance.org:cores:SweRV_EH1:1.8 --family="Cyclone V" --device=5CSEMA4U23C6 

This resulted in 198 syntax errors, most of them being something like

    Error (10170): Verilog HDL syntax error at lsu_trigger.sv(48) near text: "for";  expecting "endmodule". Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number. File: /home/olof/projects/swervolf/quartus/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/lsu/lsu_trigger.sv Line: 48
    Error (10170): Verilog HDL syntax error at lsu_trigger.sv(58) near text: "end";  expecting "endmodule". Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number. File: /home/olof/projects/swervolf/quartus/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/lsu/lsu_trigger.sv Line: 58
Error (10112): Ignored design unit "lsu_trigger" at lsu_trigger.sv(25) due to previous errors File: /home/olof/projects/swervolf/quartus/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/lsu/lsu_trigger.sv Line: 25

I think it could be that quartus doesn't like unnamed for generate loops, but I didn't investigate further.

Also, if you want to try this yourself, I found a small error in the core description file. Check out https://github.com/chipsalliance/Cores-SweRV/pull/82 for a fix

closed 2021-01-22 by aprnath