chipsalliance/Cores-VeeR-EH1 #83: How can I debug using Verilator and gdb

kingstone1927, 2020-12-02:
Hello,

I am wondering how I can debug a C program using Verilator and riscv-gdb on Core SweRV EH1. 
Do I need openOCD and JTAG to bridge Verilator and riscv-gdb? I saw people imitate a JTAG by writing a JTAG dpi, is is the only way to go? 
My goal is to debug a C program that is simulated in Verilator. I appreciate it if someone could point me to the right direction.

Thanks in advance.

agrobman, 2020-12-02:
yes, you can connect openOCD to verilator simulation, -you need to add JTAG agent, compatible with openocd, to the verilog model you simulate. However, the GDB-openocd-verilator link can be very slow. 
If your program isn't that big, you can use exec.log and *.dis file to debug what processor did, running your program.

kingstone1927, 2020-12-02:
Thank you. The *.dis seems like what I am looking for. However,  I am still curious how I can use GDB-openOCD-verilator together. 
Do you know where I can find example codes for a JTAG agent for the riscv-openocd? Also what commands or steps I need to connect opendOCD with Verilator? 
One last point: I tried writing hello_world.c as follow, but it seems like the processor does not like this. Do you know a guide that I can follow to write a proper c-program for SweRV Core? Again, Thanks so much.
```
#include <stdio.h>
int main() {
   // printf() displays the string inside quotation
   printf("Hello, World!");
   return 0;
}
```

olofk, 2020-12-02:
Hi. You can use SweRVolf for this https://github.com/chipsalliance/Cores-SweRVolf It has instructions in the readme for how to connect OpenOCD to a verilator simulator and a SoC running on FPGA

agrobman, 2020-12-02:
the DemoTB does not like your code because we don't link stdlibs and the DemoTB does not have display or/and OS with file system to print to. also you need some startup code to call main . Look at cmark.c example, how we implement printf and startup code .. Again EL2 repo has more advance support for C programs ...

JanMatCodasip, 2020-12-03:
> I am still curious how I can use GDB-openOCD-verilator together.
Do you know where I can find example codes for a JTAG agent for the riscv-openocd? Also what commands or steps I need to connect opendOCD with Verilator?

I also do recommend following @olofk's suggestion above - take a look at Cores-SweRVolf.

SweRVolf contains easy-to-follow instructions how to run SweRV core in either Verilator simulation or in an FPGA. Both the simulation and FPGA have support for UART output, so you can run programs with `printf`s and observe the output.

kingstone1927, 2020-12-03:
Thank you for your responses. This really helps. I will take a look at Core-SweRVolf

closed 2021-01-21 by aprnath