chipsalliance/Cores-VeeR-EH1 #86: Stall point

kingstone1927, 2021-01-07:
How can I know when the core is stalled when running? is there a way for me to display/know the cycle when the core is stalled?

Thanks in advance

aprnath, 2021-01-07:
Hi @kingstone1927,

SweRV's performance monitor counters allow for counting stall events. So if you want to know the internal signals for stall events, the simplest would be to look at the signals from the various blocks to the perf counter logic in the `dec` block.

Here are the signals at the `dec` level for stall events :
```
    pmu_align_stall
    ifu_pmu_fetch_stall
    dec_pmu_decode_stall
    dec_pmu_presync_stall
    dec_pmu_postsync_stall
```
Hope this helps!

kingstone1927, 2021-01-07:
@aprnath Thank you!

The simulation generates the ```exec.log``` file. Do you know if there is a way for me explicitly display "stalled cycled" on the ```exec.log```.  
I am trying to see exactly what is happening at the instruction level: what happens to the entire core when one unit is stalled, where exactly a stall happens and when it happens. Do you know if it is possible to do that and which tools I can use to do it?

aprnath, 2021-01-07:
@kingstone1927,

Yes, it is definitely possible. The `exec.log` file is created in `testbench/tb_top.v`, and all the code to monitor the commits and print the info is in that file (around line 330): https://github.com/chipsalliance/Cores-SweRV/blob/7332edc0adaa7e9a0c842d169154429e8d987786/testbench/tb_top.sv#L329
You can add any additional formatted monitors in the testbench and have them write to the `exec.log` file handle (`el`) with the appropriate timestamps to show what is happening in the pipeline.

kingstone1927, 2021-01-07:
@aprnath Wow this is really cool! Thanks!

I will take a look at the file.

aprnath, 2021-01-07:
Good luck!

closed 2021-01-07 by aprnath