chipsalliance/Cores-VeeR-EH1 #88: which unit control flushing of the pipelines?

kingstone1927, 2021-01-13:
Hello everyone,

I wonder which unit/code sections that control the flushing of the pipelines at the decode stage. 

Thank you!

agrobman, 2021-01-14:
TLU initiates pipe flush, but not in decode stage ..

kingstone1927, 2021-01-15:
@agrobman  Can you help me pointing out code sections that initiate pipe flush? 

As I understand, flush is resetting the pipelines/registers to zero values. I wonder if that is a correct way to understand flush. If so,  I am trying to find out which code sections defining the register pipelines and which code sections triggering the reset of the register pipelines. 
Please correct me if I am wrong in understanding flush. I really appreciate it if you can help me with this.

Thanks in advance.

aprnath, 2021-01-21:
I left a comment in #90

closed 2021-01-22 by aprnath