chipsalliance/Cores-VeeR-EH1 #91: No tags for releases

robtaylor, 2021-01-21:
This makes it very hard to track what is considered aa released version!

agrobman, 2021-01-21:
each release has a branch named as branch-1.X .. 
master branch is latest release

robtaylor, 2021-01-22:
I'm afraid that that doesn't help me know what the release is exactly. Also only a small number of versions have a branch.

I would suggest a standard software-style release process of bumping the version in the .core file, readme, etc, and then tagging that point with v<version number>

aprnath, 2021-01-22:
@robtaylor, actually all official releases (1.0 - 1.8) do have a branch. However your point is also valid regarding tag usage. We will  look into this - maybe even consult with a person versed in "software-style" :-)

Thanks for the suggestion.

jrahmeh, 2021-01-22:
Added tags v1.5, v1.6, v1.7, and v1.8

robtaylor, 2021-01-22:
Amazing! Thank you Ajay and jrahmeh

closed 2021-01-22 by aprnath