chipsalliance/Cores-VeeR-EH1 #92: cmark_dccm fails to build

danielmlynek, 2021-02-15:
riscv64-unknown-elf-gcc -Isnapshots/default -g -O3 -funroll-all-loops -mabi=ilp32 -march=rv32imc -nostdlib -c /builds/gitlab/gliwice/Cores-SweRV/testbench/tests/cmark_dccm/cmark_dccm.c -o cmark_dccm.o
Building cmark_dccm
riscv64-unknown-elf-gcc -mabi=ilp32 -march=rv32imc -Wl,-Map=cmark_dccm.map -lgcc -T/builds/gitlab/gliwice/Cores-SweRV/testbench/tests/cmark_dccm/cmark_dccm.ld -o cmark_dccm.exe cmark_dccm.o -nostartfiles
/opt/riscv-64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 0000000000000000
/opt/riscv-64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: cmark_dccm.o: in function `main':
/builds/gitlab/gliwice/Cores-SweRV/testbench/tests/cmark_dccm/cmark_dccm.c:1116: undefined reference to `whisperPrintf'
/opt/riscv-64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /builds/gitlab/gliwice/Cores-SweRV/testbench/tests/cmark_dccm/cmark_dccm.c:1144: undefined reference to `whisperPrintf'
/opt/riscv-64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /builds/gitlab/gliwice/Cores-SweRV/testbench/tests/cmark_dccm/cmark_dccm.c:1149: undefined reference to `whisperPrintf'
/opt/riscv-64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /builds/gitlab/gliwice/Cores-SweRV/testbench/tests/cmark_dccm/cmark_dccm.c:1060: undefined reference to `whisperPrintf'
/opt/riscv-64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /builds/gitlab/gliwice/Cores-SweRV/testbench/tests/cmark_dccm/cmark_dccm.c:1157: undefined reference to `whisperPrintf'
/opt/riscv-64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: cmark_dccm.o:/builds/gitlab/gliwice/Cores-SweRV/testbench/tests/cmark_dccm/cmark_dccm.c:1131: more undefined references to `whisperPrintf' follow

agrobman, 2021-02-15:
Daniel, check that you have latest Makefile and testbench/asm/cmark_dccm.mki - Make should compile  and link 3 files crt0.s, cmark.c and printf.c  
program compilation flow was changed a bit with 1.9

agrobman, 2021-02-15:
or just remove testbench/tests/cmark_dccm directory ..

danielmlynek, 2021-02-23:
I see that directory testbench/tests/cmark_dccm
was removed in commit: https://github.com/chipsalliance/Cores-SweRV/commit/f3da044f15dac4cea67ed3dc76fd86300f4a59f9
Update env works fine for me. Im closing this one.

closed 2021-02-23 by danielmlynek