chipsalliance/Cores-VeeR-EH1 #96: GCC version in Makefile

guqian110, 2021-05-08:
I followed the guide in README.md to run a hello world test,

    make -f $RV_ROOT/tools/Makefile verilator debug=1

the simulation finished successfully. But I find a warning message in the log:

>  !!! No riscv64-unknown-elf-gcc in path, using canned hex files !!

It seems that the prefix is wrong in the Makefile line 36 because this core only support RV32IMC according to the PRM file.

agrobman, 2021-05-08:
this warning means that that make can't run RISCV GCC. Or you don't have GCC installed or it's name isn't   riscv64-unknown-elf-gcc or it is not in the search path. You need to have the RISCV tools, supporting multilib option.
(GCC can be built to support various RISCV extensions both 32 and 64 bit versions. Makefile calls the GCC with appropriate ISA switches, suitable for EH1 . 
You can use the gcc with different prefix , supplying  GCC_PREFIX=<your riscv gcc prefix> to the make command.

agrobman, 2021-05-10:
Nothing is mandatory - you are using this code with no any guarantee ( read license statements)- it's open source project and you are getting it for free. But we advice to build 64 bit multilib RISCV tools version, which can be used with any flavor of RISCV processor. Regardless the said, you can modify everything in your local DB.

guqian110, 2021-05-11:
I got it. Although the compiled result of multilib version gcc starts with prefix `riscv64-unknow-elf`, but it does support any RV32 or RV64 cores by using different `-march`, `-abi` options, so there are not any compatible problems. Thanks for your explanation.

closed 2021-05-11 by guqian110