chipsalliance/Cores-VeeR-EH1 #97: $readmem file address beyond bounds of array

HamzaShabbir517, 2021-06-03:
Hello i was trying to load the program.hex file created from the Makefile to another memory which is declared as
reg[7:0] mem [31:0] but i am getting the error saying $readmem file address beyond bounds of array. Any possible solution for it?
Note: The testbench is written in verilog

agrobman, 2021-06-03:
looks like your memory has only 32 bytes - normally most of the tests need much more than that. Also DemoTB uses all 4GB 32 bit address space to place code and data  -  from address 0 to address 0xffffffff. TB external slaves are made around "sparse memory model", implemented as associative arrays.

HamzaShabbir517, 2021-06-03:
firstlt i have declare it like reg [31:0] mem [4095:0] but after getting error and lots of search i have convert it to the same format as the memory is define in ahb_sif.sv file. But still the error remains

agrobman, 2021-06-03:
how is the memory declared in your TB? 

if it's as in your original post, then it is defines 32 bytes memory storage, not sufficient to hold any program.
(with max address of 31)

the memory in DemoTB slaves is defined as associative array:

`bit[7:0] mem[bit[31:0]] ;`

where index is 32 bit vector.

even if you will define your memory as `reg[7:0] mem [1<<16-1:0] ` (64KB),  program.hex, built with current makefile will cause the same error, since some addresses, used in these programs are higher than 64KB.

BTW, in earlier versions of this DB we had slave memories, used static fix size memory arrays ( see earlier git branches),
which similar to RTL memories implementations( and could be synthesized into FPGA), but it placed limitation on the test program address space and made test program compilation more complicated.

You can play with test linker files, CPU memory map defines and objcopy program options to build HEX files, suitable to your TB memories sizes/widths or write ELF loader as a DPI function for verilog. 

But usage of associative arrays made test build process much simpler and more natural .

HamzaShabbir517, 2021-06-03:
What if i declare the memory as u mention reg [7:0] mem [1<<16-1:0] but instead of 16 i use 32? Will it be a possible solution. As playing with Makefile is not suitable for me as i am not musch familiar with these makefiles.

agrobman, 2021-06-03:
Your server will need > 8GB memory to simulate this...

HamzaShabbir517, 2021-06-03:
Oh ok. So can u provide any solution with respect to this solution u provided or any guide from where i can learn.
//
You can play with test linker files, CPU memory map defines and objcopy program options to build HEX files, suitable to your TB memories sizes/widths or write ELF loader as a DPI function for verilog.
//

agrobman, 2021-06-03:
gnu.org - gnu tools portal (binutils documentation - linker, objcopy, gcc , etc) 
..objcopy -h
google.com (how to ...)

HamzaShabbir517, 2021-06-03:
ok thanks

closed 2021-06-03 by HamzaShabbir517