chipsalliance/Cores-VeeR-EH1 #98: Timing violations with Vivado

Rusty-Wire, 2021-06-18:
Hello, i'm working on a softcore design based on the SweRV with Vivado.

When I proceed with the synthesis, there are a lot of timing violations problems which appears (even when running the core at 100MHz)

Any ideas about this ? Can I place some timing exceptions between registers without risks ? And which ones ? 

Maybe someone already met this situation ?

olofk, 2021-06-18:
Perhaps you would be interested in [SweRVolf](https://github.com/chipsalliance/Cores-SweRVolf), the reference platform for SweRV which has targets for simulation and implementation for a Nexys A7 board with Vivado.

It should be noted though that SweRVolf runs SweRV EH1 at 50MHz. You can speed it up a bit by playing around with the CPU configuration, e.g. reducing branch predictor, caches etc. But in practice I haven't been able to make it run faster than 65 MHz or so. Some of the constructs are highly optimized for ASIC implementation and not very suitable for FPGA which is why it won't run much faster without invasive changes to the core

Rusty-Wire, 2021-06-18:
Yes, I tried SweRVolf some time ago but it seems Vivado didn't support all the SystemVerilog fonctionnalities in simulation of some modules (don't remember which ones). I probably don't do it in the right way.

Actually, I've build up one custom design with the SweRV and SmartConnect, UART, GPIO, DDR2 MIG Xilinx IPs (All clocked at 100MHz). All is working fine in simulation and on-board (NexysA7). But yeah, my timing report is very bad.

Thank you again for the informations, I didn't know that use this CPU on an FPGA would restrict so much the target frequency.

agrobman, 2021-06-18:
Do you have RV_FPGA_OPTIMIZE define defined  for your FPGA build?
if not, you may try add -set fpga_optimize  switch to swerv.config.  it should increase maximum frequency of design by removing clock gating cells ...

Rusty-Wire, 2021-06-29:
Yeah it is in my case.

Indeed, when I stay below 50MHz, it works properly and all my timings errors disappear so thank you again !

closed 2021-06-29 by Rusty-Wire