chipsalliance/Cores-VeeR-EH1 #104: CoreMark test score

Zissi-Lei, 2021-12-07:
Hi, I achieved 5.0 at CoreMark test by running EH1 in FPGA, using master branch and the configurations you gave in the doc "SweRV_CoreMark_Benchmarking.pdf". But in your web page "https://www.westerndigital.com/solutions/business/risc-v", I see that the CoreMark score of the EH1 core is 5.7. I want to known how you configure the EH1 core to achieve 5.7 CoreMark/MHz, thanks.

aprnath, 2021-12-08:
Hi,
We get additional performance by playing with compiler flags. There are many different variations, and it is also dependent on compiler builds and versions.
Try this : 
`-mtune=sifive-7-series -mbranch-cost=1 -Ofast -funroll-all-loops -march=rv32gc -mabi=ilp32  --param=hot-bb-frequency-fraction=1 --param=max-jump-thread-duplication-stmts=2 --param=max-tail-merge-iterations=0 -fno-aggressive-loop-optimizations -fgcse-las --param=max-grow-copy-bb-insns=7 --param=unroll-jam-min-percent=0 --param=large-unit-insns=0 -fno-delete-null-pointer-checks -fno-rename-registers --param=max-partial-antic-length=1 --param=loop-max-datarefs-for-datadeps=0 -funroll-all-loops --param=uninlined-function-insns=16 -fno-tree-vrp -fwrapv -fno-toplevel-reorder --param=max-inline-insns-size=128 -fipa-pta --param=max-cse-path-length=3 --param=inline-min-speedup=1 --param=max-sched-region-insns=7 --param=max-goto-duplication-insns=0 --param=max-gcse-insertion-ratio=0 -fno-tree-ter --param=max-average-unrolled-insns=128 -fno-dce --param=max-loop-header-insns=2 -fno-tree-loop-ivcanon -fno-tree-loop-distribute-patterns -fno-sched-rank-heuristic`

There are probably other combinations that would give better results.

Zissi-Lei, 2021-12-09:
@aprnath Thank you very much! I used the compiler flags you gave above and finally achieved 6.25 CoreMark/MHz!

closed 2021-12-14 by Zissi-Lei