chipsalliance/Cores-VeeR-EH1 #108: Fusesoc's sim target is deprecated

RRozak, 2022-01-26:
When I run the following command in the root of the repository:
`fusesoc --cores-root . run --target=sim chipsalliance.org:cores:SweRV_EH1:1.8`
I get the error:
```
ERROR: Setup failed : Cannot find testbench/hex/data.hex in :
        .
        .
```
This file is required by the fileset: https://github.com/chipsalliance/Cores-SweRV/blob/master/swerv.core#L61, but it was removed from the repository in one of the previous versions.

agrobman, 2022-01-26:
I think you should remove this data.hex filename from the fileset file. It is not used anymore in simulations. Besides you should use executable build flow from tools/Makefile, instead of canned hex files.

--

RRozak, 2022-01-27:
I removed this line and changed the one with `program.hex`:
```diff
diff --git a/swerv.core b/swerv.core
index b6f986e..ef8c41e 100644
--- a/swerv.core
+++ b/swerv.core
@@ -58,8 +58,7 @@ filesets:
 
   mem_init:
     files:
-      - testbench/hex/data.hex : {copyto : data.hex}
-      - testbench/hex/program.hex : {copyto : program.hex}
+      - tools/program.hex : {copyto : program.hex}
     file_type : user
 
   tb:
```
and I got the fusesoc error:
```
INFO: Preparing chipsalliance.org:cores:SweRV_EH1:1.8
INFO: Generating chipsalliance.org:cores:SweRV_EH1-swerv_ahb_config:1.8
INFO: Wrote dependency graph to path/Documents/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/sim-verilator/chipsalliance.org_cores_SweRV_EH1_1.8.deps-after-generators.dot
INFO: Wrote Makefile fragment to path/Documents/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/sim-verilator/core-deps.mk
INFO: Setting up project
INFO: Building simulation model

Traceback (most recent call last):
  File "path/.local/bin/fusesoc", line 10, in <module>
    sys.exit(main())
  File "path/.local/lib/python3.7/site-packages/fusesoc/main.py", line 835, in main
    fusesoc(args)
  File "path/.local/lib/python3.7/site-packages/fusesoc/main.py", line 825, in fusesoc
    args.func(cm, args)
  File "path/.local/lib/python3.7/site-packages/fusesoc/main.py", line 412, in run
    args.verbose,
  File "path/.local/lib/python3.7/site-packages/fusesoc/main.py", line 515, in run_backend
    backend.build()
  File "path/.local/lib/python3.7/site-packages/edalize/edatool.py", line 187, in build
    self.build_main()
  File "path/.local/lib/python3.7/site-packages/edalize/verilator.py", line 166, in build_main
    self._run_tool('make', args, quiet=True)
  File "path/.local/lib/python3.7/site-packages/edalize/edatool.py", line 390, in _run_tool
    check=True)
  File "/usr/lib/python3.7/subprocess.py", line 467, in run
    raise ValueError('stdout and stderr arguments may not be used '
ValueError: stdout and stderr arguments may not be used with capture_output.
```

olofk, 2022-01-27:
Hi @RRozak. Thank you for reporting. I fixed this and made a PR now https://github.com/chipsalliance/Cores-SweRV/pull/109

The final Python error you encountered though is caused by something else and needs to be fixed in FuseSoC. I'm looking for more information about that right now

olofk, 2022-01-27:
@RRozak Sorry, I see now the error is in Edalize, not FuseSoC. Could you please tell me which version of Edalize you are using? If Edalize (or FuseSoC) was installed through pip, you can check with `pip3 show edalize`

RRozak, 2022-01-27:
Hi,
I have Edalize in version 0.2.4 and FuseSoC in 1.12.1.dev31+g14dfc82.

olofk, 2022-01-27:
I think you managed to hit a bug that exists in some versions of python 3.7 (https://bugs.python.org/issue36760) I can't easily reproduce it, but it might disappear if you add the `--verbose` flag, as in `fusesoc --verbose --cores-root . run --target=sim chipsalliance.org:cores:SweRV_EH1:1.8`

agrobman, 2022-01-27:
@olofk , program.hex file is used to load I and D side memories of the core TB, created by tools/Makefile prior simulation.
I don't think the file should be pre-requisite to the simulation.

@RRozak , you can try to create empty program.hex, data.hex files in testbench/hex directory to satisfy fusesoc - they are not used really

agrobman, 2022-01-27:
@RRozak ,

BTW, there is no such file:

   mem_init:
     files:
-      - testbench/hex/data.hex : {copyto : data.hex}
-      - testbench/hex/program.hex : {copyto : program.hex}
**_+      - tools/program.hex : {copyto : program.hex}_**
     file_type : user

olofk, 2022-01-27:
@RRozak The SweRV PR is merged now. If adding `--verbose` to FuseSoC doesn't solve the other problem, please open an FuseSoC bug report and we can look at it there

RRozak, 2022-01-28:
It works now, even without verbose flag. 
Thank you very much for help. I close this issue.

> BTW, there is no such file:

I my local repository there was that file, I ran make command in tools directory: https://github.com/chipsalliance/Cores-SweRV/blob/master/tools/Makefile#L163
I modified this file only to check if it can work or if another problems happen like the one with python error.

closed 2022-01-28 by RRozak