chipsalliance/Cores-VeeR-EH1 #121: Not generate waveform using vcs when testing Hello World.

zhangkanqi, 2023-12-10:
**ENV:**
vcs version:  VCS-MX2018.09-SP2
riscv64-unknown-elf-* version: 9.2.0

**PROBLEM:**
I added VCS_DEBUG into Makefile when buildng vcs:
[image]
**But when execute command `make vcs debug=1`, it didn't generate waveform.** The details are as follows:

>zhangkanqi@hwfuzz:~/Cores-VeeR-EH1/tools$ make vcs debug=1
BUILD_PATH=snapshots/default /home/zhangkanqi/Cores-VeeR-EH1/configs/veer.config -target=default -set iccm_enable
veer: Set(s) requested : iccm_enable
veer: Overriding iccm_enable value 0 with 1
VeeR configuration for target=default
veer: icache_enable                  = 1
veer: icache_tag_low                 = 6
veer: icache_size                    = 16
veer: nmi_vec                        = 0x11110000
veer: external_mem_hole              = default disabled
veer: lsu_stbuf_depth                = 8
veer: dma_buf_depth                  = 4
veer: dec_instbuf_depth              = 4
veer: fpga_optimize                  = 1
veer: lsu_num_nbload                 = 8
veer: sb_bus_tag                     = 1
veer: ifu_bus_tag                    = 3
veer: dma_bus_tag                    = 1
veer: dccm_region                    = 0xf
veer: dccm_num_banks                 = 8
veer: dccm_size                      = 64
veer: dccm_offset                    = 0x40000
veer: dccm_enable                    = 1
veer: reset_vec                      = 0x80000000
veer: ret_stack_size                 = 4
veer: pic_meigwctrl_count            = 8
veer: pic_mpiccfg_count              = 1
veer: pic_meigwclr_count             = 8
veer: pic_meipt_count                = 8
veer: pic_mpiccfg_offset             = 0x3000
veer: pic_meigwclr_mask              = 0x0
veer: pic_meipl_mask                 = 0xf
veer: pic_region                     = 0xf
veer: pic_meie_count                 = 8
veer: pic_meip_count                 = 4
veer: pic_meipl_offset               = 0x0000
veer: pic_offset                     = 0xc0000
veer: pic_meigwctrl_offset           = 0x4000
veer: pic_size                       = 32
veer: pic_total_int                  = 8
veer: pic_meigwctrl_mask             = 0x3
veer: pic_meipt_offset               = 0x3004
veer: pic_meie_offset                = 0x2000
veer: pic_meigwclr_offset            = 0x5000
veer: pic_meip_mask                  = 0x0
veer: pic_meipl_count                = 8
veer: pic_mpiccfg_mask               = 0x1
veer: pic_meip_offset                = 0x1000
veer: pic_meie_mask                  = 0x1
veer: pic_meipt_mask                 = 0x0
veer: iccm_size                      = 512
veer: iccm_num_banks                 = 8
veer: iccm_region                    = 0xe
veer: iccm_enable                    = 1
veer: iccm_offset                    = 0xe000000
veer: btb_addr_lo                    = 4
veer: btb_size                       = 32
veer: btb_index1_lo                  = 4
veer: bht_addr_lo                    = 4
veer: bht_size                       = 128
veer: Writing snapshots/default/common_defines.vh
[... 212 lines cut ...]
-rw-rw-r--  1 zhangkanqi zhangkanqi 15582870 12月 10 17:21 verilog.dump
drwxrwxr-x  5 zhangkanqi zhangkanqi     4096 12月 10 17:21 ./
-rw-rw-r--  1 zhangkanqi zhangkanqi        0 12月 10 17:21 ucli.key
-rw-rw-r--  1 zhangkanqi zhangkanqi        0 12月 10 17:21 vcs-build
-rw-rw-r--  1 zhangkanqi zhangkanqi    10980 12月 10 17:21 vcs_compile.log
drwxrwxr-x  4 zhangkanqi zhangkanqi     4096 12月 10 17:21 simv.daidir/
drwxrwxr-x  6 zhangkanqi zhangkanqi     4096 12月 10 17:21 csrc/
-rwxrwxr-x  1 zhangkanqi zhangkanqi   841312 12月 10 17:21 simv*
-rw-rw-r--  1 zhangkanqi zhangkanqi     1086 12月 10 17:21 vc_hdrs.h
-rw-rw-r--  1 zhangkanqi zhangkanqi     4088 12月 10 17:21 hello_world.dis
-rw-rw-r--  1 zhangkanqi zhangkanqi     1763 12月 10 17:21 hello_world.map
-rwxrwxr-x  1 zhangkanqi zhangkanqi     1113 12月 10 17:21 program.hex*
-rwxrwxr-x  1 zhangkanqi zhangkanqi     8876 12月 10 17:21 hello_world.exe*
-rw-rw-r--  1 zhangkanqi zhangkanqi     1136 12月 10 17:21 hello_world.o
-rw-rw-r--  1 zhangkanqi zhangkanqi      969 12月 10 17:21 hello_world.cpp.s
drwxrwxr-x  3 zhangkanqi zhangkanqi     4096 12月 10 17:21 snapshots/
drwxrwxr-x 12 zhangkanqi zhangkanqi     4096 12月 10 17:12 ../
-rwxrwxr-x  1 zhangkanqi zhangkanqi     6411 12月 10 16:37 Makefile*
-rwxrwxr-x  1 zhangkanqi zhangkanqi     1369 12月  8 17:05 addassign*
-rwxrwxr-x  1 zhangkanqi zhangkanqi     5550 12月  8 17:05 coredecode*
-rw-rw-r--  1 zhangkanqi zhangkanqi    67705 12月  8 17:05 JSON.pm
-rwxrwxr-x  1 zhangkanqi zhangkanqi     1877 12月  8 17:05 picmap*
-rwxrwxr-x  1 zhangkanqi zhangkanqi     2884 12月  8 17:05 smalldiv*
-rwxrwxr-x  1 zhangkanqi zhangkanqi    10260 12月  8 17:05 unrollforverilator*
-rw-rw-r--  1 zhangkanqi zhangkanqi       73 12月  8 17:05 vivado.tcl

algrobman, 2023-12-10:
verilog.dump is your VCD waves

zhangkanqi, 2023-12-11:
But it doesn't have source code, i can not trace code by clicking.

[verilog.dump.zip.zip](https://github.com/chipsalliance/Cores-VeeR-EH1/files/13630295/verilog.dump.zip.zip)