chipsalliance/Cores-VeeR-EH1 #125: extension `zicsr' required

ahmdotm, 2024-06-19:
I used --with-arch=rv64imac_zicsr --with-abi=lp64 for my toolchain configuration and I still get this:
 ized opcode `csrr a6,mcycleh', extension `zicsr' required
make: *** [/home/othman/Reev_EH1/Cores-VeeR-EH1/tools/Makefile:182: printf.o] Error 1

noobwei, 2024-12-25:
I don't know if it's accurate but a similar issue happened to me. 

I used gcc in https://github.com/riscv-collab/riscv-gnu-toolchain and changed the prefix to  `/opt/riscv/bin/riscv64-unknown-linux-gnu`
Then I got  `Error: unrecognized opcode csrw 0x7c0,x1, extension zicsr required`

I didn't succeed in adding `ziscr` to the default Makefile. It didn't make sense to simply say `ABI = -mabi=ilp32 -march=rv32imc_zicsr`, because I got `/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/../../../../riscv64-unknown-linux-gnu/lib/libgcc_s.so.1: error adding symbols: file in wrong format'`

Merry Christmas to you all btw.

noobwei, 2024-12-26:
I have fixed my issue by rebuilding the toolchain and using `--with-arch=rv32imc_zicsr --with-abi=ilp32`.

JackySu, 2025-02-26:
> I have fixed my issue by rebuilding the toolchain and using `--with-arch=rv32imc_zicsr --with-abi=ilp32`.

Can't thank you enough for this, besides make sure to modify tools/Makefile
1. L23: `ABI = -mabi=ilp32 -march=rv32imc_zicsr`
2. L77: `CFLAGS += "-std=c++17"`