chipsalliance/Cores-VeeR-EH1 #128: Hit max cycle count (10000000) .. stopping

noobwei, 2025-01-08:
This issue continues some issues from the previous issue: [https://github.com/chipsalliance/Cores-VeeR-EH1/issues/125](url)
My colleague is configuring the environment in a new VM according to the documentation wirtten by me: [https://github.com/noobwei/Y-Lab/blob/main/20241226.md](url)

But build the tool chain without the tags like `--with-arch=rv32imc_zicsr --with-abi=ilp32`, but a riscv64 toolchain.

Here is the log of `riscv64-unknown-elf-gcc -v`:

```shell
wjy1@wjy1:/opt/riscv/bin$ export PATH=/opt/riscv/bin:$PATH
wjy1@wjy1:/opt/riscv/bin$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-elf/14.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/wjy1/桌面/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.2.0 ()
```

So I modified the L23 of Makefile to  `--with-abi=lp64d_zicsr --with-arch=rv64gc`

But the result of `make -f $RV_ROOT/tools/Makefile` is `Hit max cycle count (10000000) .. stopping`

Here is the full log:

```shell
wjy1@wjy1:~/桌面/Cores-VeeR-EH1$ make -f $RV_ROOT/tools/Makefile
rm -rf *.log *.s *.hex *.dis *.tbl irun* vcs* simv* *.map snapshots veer* \
verilator* *.exe obj* *.o ucli.key vc_hdrs.h csrc *.csv \
work dataset.asdb  library.cfg
BUILD_PATH=snapshots/default /home/wjy1/桌面/Cores-VeeR-EH1/configs/veer.config -target=default -set iccm_enable
veer: Set(s) requested : iccm_enable
veer: Overriding iccm_enable value 0 with 1

VeeR configuration for target=default

veer: bht_size                       = 128
veer: bht_addr_lo                    = 4
veer: reset_vec                      = 0x80000000
veer: sb_bus_tag                     = 1
veer: ifu_bus_tag                    = 3
veer: dma_bus_tag                    = 1
veer: btb_index1_lo                  = 4
veer: btb_addr_lo                    = 4
veer: btb_size                       = 32
veer: icache_enable                  = 1
veer: icache_size                    = 16
veer: icache_tag_low                 = 6
veer: pic_meie_count                 = 8
veer: pic_meigwctrl_offset           = 0x4000
veer: pic_mpiccfg_count              = 1
veer: pic_meigwclr_offset            = 0x5000
veer: pic_total_int                  = 8
veer: pic_meipt_offset               = 0x3004
veer: pic_meip_count                 = 4
veer: pic_size                       = 32
veer: pic_meipl_offset               = 0x0000
veer: pic_mpiccfg_offset             = 0x3000
veer: pic_offset                     = 0xc0000
veer: pic_region                     = 0xf
veer: pic_meigwclr_count             = 8
veer: pic_meigwctrl_mask             = 0x3
veer: pic_meipt_count                = 8
veer: pic_meie_mask                  = 0x1
veer: pic_meip_offset                = 0x1000
veer: pic_meigwctrl_count            = 8
veer: pic_meipl_mask                 = 0xf
veer: pic_meipl_count                = 8
veer: pic_meipt_mask                 = 0x0
[... 65 lines cut ...]
ccache g++   -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0-DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++14   -include Vtb_top__pch.h.slow -c -o Vtb_top___024unit__DepSet_hba4cee98__0__Slow.o Vtb_top___024unit__DepSet_hba4cee98__0__Slow.cpp
ccache g++   -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0-DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++14   -include Vtb_top__pch.h.slow -c -o Vtb_top__Syms.o Vtb_top__Syms.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++14   -include Vtb_top__pch.h.fast -c -o Vtb_top.o Vtb_top.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++14   -include Vtb_top__pch.h.fast -c -o Vtb_top___024root__DepSet_h7f86b803__0.o Vtb_top___024root__DepSet_h7f86b803__0.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++14   -include Vtb_top__pch.h.fast -c -o Vtb_top___024root__DepSet_h9ad02603__0.o Vtb_top___024root__DepSet_h9ad02603__0.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++14   -include Vtb_top__pch.h.fast -c -o Vtb_top___024root__DepSet_h9ad02603__1.o Vtb_top___024root__DepSet_h9ad02603__1.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++14   -include Vtb_top__pch.h.fast -c -o Vtb_top___024root__DepSet_h9ad02603__2.o Vtb_top___024root__DepSet_h9ad02603__2.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++14   -include Vtb_top__pch.h.fast -c -o Vtb_top___024root__DepSet_h9ad02603__3.o Vtb_top___024root__DepSet_h9ad02603__3.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++14   -include Vtb_top__pch.h.fast -c -o Vtb_top___024root__DepSet_h9ad02603__4.o Vtb_ [... cut ...] o Vtb_top___024root__DepSet_h9ad02603__6.o Vtb_top___024root__DepSet_h9ad02603__6.cpp
g++    test_tb_top.o verilated.o verilated_threads.o Vtb_top__ALL.a    -pthread -lpthread -latomic   -o Vtb_top
rm Vtb_top__ALL.verilator_deplist.tmp
make[1]: 离开目录“/home/wjy1/桌面/Cores-VeeR-EH1/obj_dir”
touch verilator-build
./obj_dir/Vtb_top

VerilatorTB: Start of sim

Hit max cycle count (10000000) .. stopping
- /home/wjy1/桌面/Cores-VeeR-EH1/testbench/tb_top.sv:332: Verilog $finish

VerilatorTB: End of sim

```

zhishuohuang9111, 2025-03-25:
Have you done your problems?

noobwei, 2025-03-26:
> Have you done your problems?

Not solved yet
I posted this issue for my colleague, and this problem did not occur on my own machine.
She may not be following up now, so it may not be reproducible

zhishuohuang9111, 2025-03-26:
Sometimes when I changed the codes and run coremarks, I also meet this problems: Hit max cycle count (10000000) .. stopping